wiedehopf/tar1090 #2: Docker image

kevinelliott, 2019-09-12:
Hey @wiedehopf ... this isn't an issue, but I didn't know how to get ahold of you.

I wanted to let you know that I made a docker config and image of this project so that it can be run in docker alongside dump1090/skyview1090.

https://github.com/boxelio/dockerfiles/tree/master/wiedehopf-tar1090

It's part of my [Boxel dockerfiles](https://github.com/boxelio/dockerfiles) project, which is currently a collection of radio related apps (though it won't be only radio related eventually) that can run in containers.

[image]

Feel free to reach me on the ADBS Exchange Discord (as KevinE) or here.

Thanks for making this dump1090 fork! I like it a lot and use it next to skyview1090.

wiedehopf, 2019-09-12:
I don't particularly like containers, but go ahead, that's just personal preference!

Anyway the project is missing the graphs: https://github.com/wiedehopf/graphs1090#graphs1090
Just saying :)

kevinelliott, 2019-09-12:
Understood. Just wanted to let you know about it.

The purpose of this is for easily/quickly preparing a highly optimized Raspberry Pi by running an ultra-slim host OS and isolating various radio apps into containers, and letting users customize what is running on their pi dynamically... reducing the difficulty for getting customized installations going on their pis. Eventually I'll have a management tool letting users turn on/off various components during install or while running... and of course make upgrading components easier. This becomes apparent when you are running 10-15 different things on a single host and don't remember all of the steps necessary 6 months later.

Cool, I'll add the graphs too. Was planning on it already.

commit 64bbaa1c (2025-09-21): MergeNonIcao fix #2
closed 2019-09-13 by wiedehopf