wiedehopf/tar1090 #3: Manual installation

Deliaz, 2019-10-03:
Thanks for the project.
I am wondering, is there a docs how to install it manually? 
I have my own configuration: dump1090-mutability + fr24 + piaware + planefinder. So I cannot relly to dump-fa folder, I don't have it.

wiedehopf, 2019-10-03:
The dump1090 folder should be autodetected since some time.
Not all data will be available with mutability, but most will.

In regards to nginx, you'll have to figure that out yourself.
It's somewhat complicated though due to the header stuff in the lighttpd config.

Deliaz, 2019-10-04:
Ok, thanks.

I made it work with mutability. It didn't detect the folder automatically, so I customized the configs.
Maybe I will make PR to improve this condition: https://github.com/wiedehopf/tar1090/blob/master/install.sh#L53

Also, I've switched to lighttpd.
Btw, I thought that mutability provides more data than other builds.

wiedehopf, 2019-10-04:
It's also possible to give the source directory as an argument for the installation.
(https://github.com/wiedehopf/tar1090#installation--update-to-work-with-another-folder-for-example-runcombine1090)

I'm not sure why the condition wouldn't work, is the the /run folder not /run/dump1090?
If it's /run/dump1090-mutability i can just add that to the folders being checked but i thought it was /run/dump1090.

Deliaz, 2019-10-04:
You are right. It is `/run/dump1090-mutability`. Condition works, it just does not support such case.

wiedehopf, 2019-10-04:
So i'm not right, the case in the script was meant for dump1090-mutability.

dump1090-fa is the continuation of dump1090-mutability development, same guy.
It's just some stuff such as IAS, TAS and selected altitude and stuff like that.
Not always transmitted depending on difference what ATC radar queries the transponders.

Deliaz, 2019-10-04:
Thanks for the info. 
I used mutability, because not sure if dump1090-fa compatible with fr24 and planefinder.

wiedehopf, 2019-10-04:
The beast output is identical, which is what should be used for fr24 and planefinder.
(Just if you are curious: https://github.com/wiedehopf/adsb-wiki/wiki/Installing-dump1090-fa)

Anyway i've fixed the condition to auto-detect dump1090-mutability :)
I can ask someone who made it work with nginx, if you are still interested in that.

Deliaz, 2019-10-04:
Thanks for the link, will check.

Configuring nginx is not a problem, I just need to find some free time to do this. 
But lighttpd also suits me and already works fine.

wiedehopf, 2019-10-04:
Yeah i do some header magic to make the browser decompress the gz files, which are compressed by my script and not the webserver.
So the webserver needs to add some custom headers, see the lighttpd config:
```
 setenv.add-response-header += (
         "Content-Encoding" => "gzip",
         "Content-Type" => "application/json",
```

closed 2019-10-11 by wiedehopf