wiedehopf/tar1090 #7: Unable to click on field data (feature request)

memedude, 2020-02-12:
When I am viewing an aircraft and click on the aircraft to display the left side information panel I would like to be able to click on the Reg or ICAO (hex) but as soon as I double-click on something it de-selects. I assume that the problem is caused by a refresh of data. I looked at the /usr/local/share/tar1090/html/config.js file for an option to delay panel refresh, but I could not find anything there.  Is this a configurable option?

wiedehopf, 2020-02-16:
refresh depends on the --write-json-every setting configured in dump1090/readsb
that's not a good approach though.

i'll work on changing the fields only if they have changed.
it'll be a bit though.

memedude, 2020-04-01:
just installed the update and the solution you added works great. Thank You!