wiedehopf/tar1090 #8: tar1090 loading but with no data or map

ilovepancakes95, 2020-02-18:
I have a working install of dump1090_sdrplus that I use with a Hack RF One, running on Ubuntu. I want to use tar1090 as the web interface. I install it and follow the directions to setup and add config to nginx, however, the tar1090 web interface loads at http://localhost/tar1090 but there is no data displayed, and not even an empty map, just a blank white part of the page where map should go. I am guessing it has something to do with the link between dump1090 and tar1090. Below is my nginx default config, where I did change the "/run/dump1090-fa" line to the correct path where my dump1090 fork is running. The instructions seems to mention only changing this value if using a fork other than dump1090-fa. Included a screenshot below too.

nginx config: https://pastebin.com/kqssXhDY

[image]

wiedehopf, 2020-02-18:
Use this and install the package:
 https://github.com/wiedehopf/adsb-wiki/wiki/Building-readsb-from-source

Then edit /etc/default/readsb and replace the contents with this:
```
RECEIVER_OPTIONS="--net-only --net-connector localhost,30005,beast_in"
DECODER_OPTIONS="--lat 50.0 --lon 10.0 --max-range 320"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1200 --net-ro-interval 0.1 --net-ri-port 0 --net-ro-port 0 --net-sbs-port 0 --net-bi-port 40004 --net-bo-port 0"
JSON_OPTIONS="--json-location-accuracy 2"
```

If you want mlat results, you'll have to modify it the mlat configuration to feed results to 40004.

---

If you want to use the existing dump1090 to create the json files, some properties like TAS/IAS will be missing.
To make your existing install work you will probaly need this:
```
--write-json /run/dump1090
```
you will also need to create that folder via the systemd service or by other means.

This is what a typical readsb systemd unit file looks like just for your inspiration:
```
root@x230 ~ # cat /lib/systemd/system/readsb.service 
# readsb service for systemd

[Unit]
Description=readsb ADS-B receiver
Documentation=https://github.com/mictronics/readsb
Wants=network.target
After=network.target

[Service]
EnvironmentFile=/etc/default/readsb
User=readsb
RuntimeDirectory=readsb
RuntimeDirectoryMode=0755
ExecStart=/usr/bin/readsb \
 $RECEIVER_OPTIONS $DECODER_OPTIONS $NET_OPTIONS $JSON_OPTIONS \
 --write-json /run/readsb --quiet
Type=simple
Restart=on-failure
RestartSec=30
Nice=-5

[Install]
WantedBy=default.target
```

wiedehopf, 2020-02-18:
whatever you do, check that the aircraft.json and receiver.json in the /run directory you have in the nginx configuration.

ilovepancakes95, 2020-02-19:
I'll try your above steps, but yeah there is definitely no .json files being made by dump1090. And if I try to use the --write-json argument, it doesn't seem to exist in this fork for some reason.... it doesn't show up on --help either.

wiedehopf, 2020-02-19:
if it has the integrated webserver, that's not compatible with tar1090 and you'll have to use readsb with net-only mode as described.

wiedehopf, 2020-02-19:
Note that you only need the steps above the line to make the --net-only solution work.

ilovepancakes95, 2020-02-19:
For readadsb I should do the regular install or adventurous one? Not adventurous with software like this yet unless it is necessary.

wiedehopf, 2020-02-19:
It probably doesn't matter, the regular should do just fine!

ilovepancakes95, 2020-02-20:
I installed readsb but upon starting it (when using --net-only too) I get no supported devices are found. If I understand correctly, I am supposed to use readsb in place of dump1090 I have? Either way I tried with both running and still nothing gets piped into tar1090.

wiedehopf, 2020-02-20:
I gave you a complete configuration file for readsb, if you can't use it then that's not my issue.

Once readsb is working and /run/readsb has data, just reinstall tar1090 and it will use that directory.

No readsb doesn't replace the dump1090 fork you are using, which --net-only implies:
It's only running with network input.

ilovepancakes95, 2020-02-20:
OK, that's what I was missing, the correct path where readsb makes the jsons. So I updated the nginx config to point /tar1090/data to /run/readsb and confirmed that readsb is indeed putting jsons there. So now when I load tar1090 webpage it actually loads the map, but no aircraft are displayed on that or the list, it says 0. I rebooted the machine too to get all services to restart cleanly. Also tried a re-install of tar1090 using that command to point to a different run directory, but same results as above. the nginx-tar1090.conf shows /run/readsb as the directory, which is where jsons are being updated now.

wiedehopf, 2020-02-20:
are you actually receiving anything?

in the webbrowser look at this:
/tar1090/data/aircraft.json
/tar1090/data/receiver.json

if you have those, it's working.

If you have readsb connecting correctly (check `sudo journalctl -u readsb`) and you still have no aircraft, there is an issue with your reception or the original dump1090.
I can't help you with that.

Just buy a Raspberry Pi and one of the sticks outlined here: 
https://github.com/wiedehopf/adsb-wiki/wiki/Useful-items-for-purchase

closed 2020-02-20 by wiedehopf

ilovepancakes95, 2020-02-20:
I am definitely getting reception and I can see aircraft on the --interactive mode of dump1090. I am also running fr24feed which is successfully picking up data on port 30002 out of dump1090. I noticed your sample config for readsb and the output of journalctl -u readsb pointed to 30005 being used for input, so I changed that config to 30002 and restarted. The json files have contents in them. The aircraft one says 0 for messages though and the receiver one is counting up on the history of aircraft identified. Still no aircraft showing up unfortunately even after changing port to 30002 which dump1090 is set to use. Thanks for the input.

wiedehopf, 2020-02-20:
does that dump1090 fork even have beast output?

if it only puts out AVR, you're out of luck.

wiedehopf, 2020-02-20:
Well actually  ... you can change the beast_in to raw_in and it should work.

I just don't like dealing with anything but beast.

ilovepancakes95, 2020-02-20:
I guess it is AVR only? The two ports it outputs on are 30002 and 30003.

On 30002 an example messages I see is: *8DAC82F8EA428866D13C088E17DC;

On 30003 an example output showing up is: MSG,3,,,A443D3,,,,,,,12525,,,40.00388,-75.40680,,,0,0,0,0

So as you suggested I changed it to raw_in and used 30002 and the aircraft finally show up in tar1090, EXCEPT with no position data, so they only show up on list, not map. Clearly that 30003 output has position data in it, but can't make sense of the 30002 one, not familiar with the AVR outputs.

wiedehopf, 2020-02-20:
you need to put in the correct position.

with the configured max-range .... it filters all planes.

ilovepancakes95, 2020-02-20:
And, it all seems to work now! Thanks again, for your help and input. It's a great piece of software you made.

wiedehopf, 2020-02-20:
next time you have such an issue, try reading the help output for the programs more.

if you check data input and output, it becomes clear faster what the issue is.

anyhow, good day!