wiedehopf/tar1090 #10: Aircraft Ownership (Feature Request)

viksra, 2020-02-23:
Requesting aircraft owner's name/company is displayed again.

jpaladino, 2020-06-02:
This would be a huge improvement.

zyphlar, 2020-07-21:
Simply allowing users to click the callsign and go to `https://registry.faa.gov/aircraftinquiry/NNum_Results.aspx?NNumbertxt={registration_number}` (like https://registry.faa.gov/aircraftinquiry/NNum_Results.aspx?NNumbertxt=N433DF for N433DF) would save a bunch of clicking without having to integrate an API.

Similarly, searching Jetphotos with one click would also be great: `https://www.jetphotos.com/photo/keyword/{registration_number}` (like https://www.jetphotos.com/photo/keyword/N433DF )

Drawer, 2020-08-14:
It's even easier to get the picture of a plane via the API of airport-data.com:
https://www.airport-data.com/api/doc.php
Maybe not as complete as jetphotos, but easier to implement I'd guess.

viksra, 2020-08-30:
I hope the community doesn't mind me asking, but I posted this in February and this is still not done. With the solutions offered by participants here, why hasn't this been done yet? It does not seem like a difficult "fix" given that it existed before.

wiedehopf, 2020-08-30:
I might add a planespotters link i suppose.

No one submitted a pull request, those i do consider more closely.
This is a personal project and i don't feel obligated to add features.

Often i will only leave the issue open because it makes no sense for other people to open the same issue.

zyphlar, 2020-08-30:
I will note that I appreciate the flightaware links in this package even if
my main use case, browsing adsbx, doesn't enable them. Maybe I'll just
throw together a quick userscript to inject the links client side.

closed 2020-08-31 by wiedehopf

wiedehopf, 2020-08-31:
Added the planespotters link:
https://github.com/wiedehopf/tar1090/commit/9b5f1d61d28409bfdd23ca8f94874cd00ac1d63f

slim-bean, 2020-09-18:
> No one submitted a pull request, those i do consider more closely.
> This is a personal project and i don't feel obligated to add features.

I can empathize with you here.

I just want you to know I love this project and appreciate and thank you for everything you do @wiedehopf !

viksra, 2021-03-04:
Just leaving a comment to say thank you to @wiedehopf for adding back in the ownership information once again. It is very useful to have it there instead of having to Google the tail number for each plane.