wiedehopf/tar1090 #14: Persistent filters (Feature Request)

sxdjt, 2020-03-02:
It would be great it we could set filters and have them persist from one session to the next.

Use case/example:  I don't care about aircraft on the ground, so I set an altitude filter to exclude anything under 10 feet.  

The filter works as expected, but I have to set it every time I view the site.

rin67630, 2020-03-11:
+1

additionally it would probably make sense to erase tracks on map each time a filter is set: currently the tracks that are on the map before the filter went in operation stay forever on the map.

wiedehopf, 2020-03-11:
The track will only stay on the map until the aircraft is deselected.

But seems in case you have all trails (all aircraft selected) enabled, that case isn't properly handled.

PR #152 (2021-10-02): Add a setting to persist filters to local storage.
This will add a setting (disabled by default) to persist the filters to local storage.  

This should resolve #14

I tested with all tracks enabled and then enabled a filter and it cleared all non selected plans that don't match.
closed 2021-10-02 by wiedehopf