wiedehopf/tar1090 #29: Change the source ip

axlenick, 2020-03-25:
Is there a config file I can change to pull adsb data from virtual radar since I have multiple feeds?

wiedehopf, 2020-03-25:
That's not what tar1090 is ... it's just an interface for dump1090-fa or readsb providing the JSON files.

You could install multiple readsb instances, use --net-connector to get data from somwhere for each instance, then create multiple tar1090 instances to use the /run folders you have chosen for the readsb instances.

But you'll have to figure that out yourself.

wiedehopf, 2020-03-25:
https://github.com/wiedehopf/tar1090#multiple-instances

https://github.com/wiedehopf/adsb-wiki/wiki/Building-readsb-from-source

readsb --help

For multiple instances, it's best to understand how a systemd service works:
https://github.com/wiedehopf/adsb-wiki/wiki/Generic-systemd-service

You'll need to copy and modify the readsb systemd file though as it needs the RunFolder and some other stuff.

readsb gets the data from the network with `--net-only --net-connector 192.168.2.3,30005,beast_input`
depending on the --json-dir option it will write that to a directory, the default service uses `/run/readsb`
tar1090 is then set up to use data from `/run/readsb`

closed 2020-03-25 by wiedehopf