wiedehopf/tar1090 #30: (F)ollow Button Not Working

Aberdon, 2020-03-26:
Currently using tar1090 on the adsbexcehange.com website and the '(F)ollow' button is not working, when an aircraft is manually selected.  It worked 2 days ago, but yesterday and today it no longer works.

If you 'Follow a (R)andom plane' then the '(F)ollow' button becomes active.  If you then deselect that aircraft the '(F)ollow' button stays active even though there is no longer any aircraft selected.  

If, after you 'Follow a (R)andom plane' you then select a different aircraft the '(F)ollow' button becomes inactive, but you can not follow the newly selected aircraft.

If after you 'Follow a (R)andom plane' you then select 'Follow a (R)andom plane' again, a different aircraft is selected and the '(F)ollow' button is shown as active.

wiedehopf, 2020-03-26:
Using the keyboard for it was still working, that's why i didn't notice.

Should be fixed :)

closed 2020-03-26 by wiedehopf

Aberdon, 2020-03-26:
Thanks wiedehopf - now working