wiedehopf/tar1090 #32: Offline capable maps (Feature Request)

chjohn22, 2020-04-30:
As the development of [jprochazkas adsb-receiver](https://github.com/jprochazka/adsb-receiver) stopped, the long requested feature to include offline capable maps in the webinterface was not developed.
Would it be possible to implement this feature in tar1090? For example to render extracts from OSM in .pfb-format.
I do not know how difficult such an implementation would be, but modern tools like [Mapnik](https://mapnik.org/) and [OSM Tile Server](https://wiki.debian.org/OSM/tileserver/buster) seem to be running well.

zenonp, 2020-09-27:
Take a look at [osmand-tools](https://github.com/osmandapp/OsmAnd-tools). It generates offline maps in obf format and it's GPLv3-licensed. It could be incorporated into tar1090 (making things easy at the cost of some bloat), or it could be used by the end-user if tar1090 provides a hook (a bit more hassle, but with the opportunity to make one's own styling and layers).

wiedehopf, 2022-06-13:
It's also possible to make the map move with gpsd if you use readsb as the decoder.
readsb command line option to be added to /etc/default/readsb: --net-connector localhost,2947,gpsd_in

novwhisky, 2023-02-22:
That link now points to an empty gh-pages branch. Is there a better reference?

wiedehopf, 2023-02-23:
I don't know how long the referenced files will remain online: https://github.com/ogadsb/wiki/wiki/tar1090-offline-map

Also as i don't have plans to update those files, the maps will become outdated.

trevelyanuk, 2023-03-07:
Would there be any merit or use in users being able to select or configure a tile source in the UI, or for it to be able to be configured from command line? Then users could have the tile source being locally hosted, web hosted or somewhere else on the local network. 

(I also take on board even if that's a "yes", it doesn't mean you are going to be able to do it 😂)

zenonp, 2023-03-08:
> Would there be any merit or use in users being able to select or configure a tile source in the UI

Check https://github.com/traccar/traccar-web . Its web interface offers 10 different map sources (osm, yandex, bing etc) and more can easily be added to the back-end. Thus: the code is already written, it's GPL, it can be just copied and reused.

addyh, 2023-04-10:
> I don't know how long the referenced files will remain online: https://github.com/ogadsb/wiki/wiki/tar1090-offline-map
> 
> Also as i don't have plans to update those files, the maps will become outdated.

The tile files are taken down from adsbexchange site... Can you post a new link maybe?

chjohn22, 2023-05-13:
@addyh @wiedehopf Maybe someone can extract the files from his Pi and upload them somewhere? I could host them, but I do not have the files anymore, my SD card died...

wiedehopf, 2023-05-13:
https://github.com/wiedehopf/adsb-wiki/wiki/offline-map-tiles-tar1090

closed 2023-05-14 by chjohn22