wiedehopf/tar1090 #33: How to start tar1090 with readsb ?

fe8769a, 2020-05-03:
How to start tar1090 with readsb ?

hello

I try to run tar1090 with readsb . Readsb is working fine with virtual radar server in mono mode .

As I run http://127.0.0.1/tar1090 : system said : impossible to start connection 

sudo systemctl restart tar1090 is running with no error

I fill the file :

sudo nano /etc/default/tar1090_instances with 
 
/run/dump1090-fa tar1090
/run/dump1090-fa persist

I checked the /run directories , they aren't empty

What can I check to see if web server is running ? What can I check to make tar1090 ri=unning

Thanks

wiedehopf, 2020-05-03:
Check the lighttpd logs.
https://github.com/wiedehopf/adsb-wiki/wiki/Debug-commands

127.0.0.1 ? is that the machine you run readsb/tar1090 on?
You'll need the address for that machine.

wiedehopf, 2020-05-03:
Oh just saw that now.

/run/dump1090-fa is not the right directory for readsb ... (/run/readsb)

closed 2020-05-05 by wiedehopf