wiedehopf/tar1090 #37: Filter out "planes" with 1 or few messages

giacomo892, 2020-05-06:
I cannot find a way to do not show planes that has 1 message received at most. 
They are likely to be just receiver errors.
Like this: 
[image]
The name starts with _ too.

wiedehopf, 2020-05-06:
That shouldn't be shown with a proper decoder like dump1090-fa or readsb.
It's sometimes important to show everything the decoder will putin the json.

giacomo892, 2020-05-06:
Hi, thanks for your reply.
I am using readsb right now

wiedehopf, 2020-05-06:
Which version are you using?
I've just changed it for my dev branch, but generally that branch .... well does what i want it to do.
Don't expect that branch to be reasonable ;)

giacomo892, 2020-05-06:
I'm using readsb 3.8.1 and tar1090 installed yesterday

wiedehopf, 2020-05-06:
Could you elaborate on which readsb you installed exactly?

giacomo892, 2020-05-06:
https://github.com/Mictronics/readsb

wiedehopf, 2020-05-06:
3.8.3 isn't the current version.

Anyhow, aircraft with only one message shouldn't be in the json.

```
            if (a->messages < 2) { // basic filter for bad decodes
                continue;
            }
```

 Are you sure you aren't using a strange branch or sth ...

wiedehopf, 2020-05-06:
correction, 3.8.1 isn't the current version 3.8.3 is.
But that's besides the point, i don't think this code was ever different.

giacomo892, 2020-05-06:
I've just upgraded, idk why I was running 3.8.1
Will report. thanks!

giacomo892, 2020-05-06:
Fine now, thanks!

closed 2020-05-06 by giacomo892

wiedehopf, 2020-05-06:
just out of curiosity, let's check the output for this command:

readsb --version

giacomo892, 2020-05-06:
readsb --version
Mictronics v3.8.3

giacomo892, 2020-05-06:
Should I run your fork?

wiedehopf, 2020-05-06:
No you shouldn't.

Just puzzled how that 1 message plane made it into there.

giacomo892, 2020-05-06:
:)

giacomo892, 2020-05-12:
Continuing here so you already know my setup.
I've a strage issue. I've cleaned the cache and everything...
On Chrome I cannot see MLAT traffic, it just shows are Mode S
on the very same chrome, but incognito mode, it shows in yellow and as MLAT with position.

Any clue?

thanks

wiedehopf, 2020-05-12:
Shift-N

giacomo892, 2020-05-12:
Will try as soon I have traffic :D Thanks