wiedehopf/tar1090 #45: Cache housekeeping?

giddyhup, 2020-06-16:
My Raspberry Pi's file system ran out of inodes. I noticed that /var/cache/lighttpd/compress/tar1090 is filling up rapidly. Is there something I can do to prevent this?

wiedehopf, 2020-06-17:
That's not my issue because the default behaviour doesn't cache compressed json in lighttpd.

Anyhow remove the cache dir line from your lighttpd config and it will stop caching and instead compress on the fly.

closed 2020-06-17 by wiedehopf

giddyhup, 2020-06-17:
@wiedehopf, please accept my honest apology for blaming tar1090 for causing this. Apparently, another feeder/gui activated the compression mod and that mod's default 30 day cleanup policy and default settings do not go well together with the dynamic content delivered.

wiedehopf, 2020-06-17:
Just remove the cache dir line from the lighttpd config and you'll be fine.

No problem, have a nice day!