wiedehopf/tar1090 #49: Aircraft icon size

BAZZZERH, 2020-07-12:
I have been trying to way of reducing the aircraft icon defaut size as when viewing on a small screen they are a tad too big... could you point me as to where the default size line is please...

wiedehopf, 2020-07-13:
Did you try the + toggle in the top left?

You can change the config.js
https://github.com/wiedehopf/tar1090#configuring-the-web-interface-optional

BAZZZERH, 2020-07-13:
Tried the + toggle, but the smallest size is still too large... the original fa1090 size is a bit smaller which is better for a smaller screen.. . Also I have tried various settings in the config and not found anything that makes the plane icons smaller... if someone can suggest the line to change I would appreciate it....

wiedehopf, 2020-07-13:
```
// -- Marker settings -------------------------------------

// different marker size depending on zoom lvl
// markerZoomDivide = 8.5;
// marker size when the zoom level is less than markerZoomDivide
// markerSmall = 1;
// marker size when the zoom level is more than markerZoomDivide
// markerBig = 1.18;
```

closed 2020-07-13 by wiedehopf