wiedehopf/tar1090 #62: Aircraft type/type code no longer shown

gtjoseph, 2020-08-10:
In the current master branch (d00cc809a5a1a76d75ef1e9526787fadacc2673b), the type/type code is blank.  It was working with commit 118891ca9cae5cb922b8657ed222350e00b327e2 from July 14th (the last time I pulled).   I'll see if I can bisect and fix later today if I can find the time but I just wanted to mention it in case you're already working on it.

wiedehopf, 2020-08-10:
It works fine for me.

Likely a caching issue, how do you install?

gtjoseph, 2020-08-10:
Cleared cache and cookie/local storage with no change.  I had just copied the html directory to /usr/local/share/tar1090/html but I just ran the install script and for the same issue.  However, I just tried Firefox and it's working.   I'm running Chrome 
"Version 85.0.4183.59 (Official Build) beta (64-bit)".  I guess something changed.

wiedehopf, 2020-08-10:
Just copying the html is not gonna work.
The database is in an extra repo, it's required:
https://github.com/wiedehopf/tar1090-db
That db folder in that repo needs to be in /html

Anyhow i've received no other complaints, pretty sure you're doing something wrong.
(maybe didn't clear cache after running the install script)

closed 2020-08-10 by wiedehopf