wiedehopf/tar1090 #64: Play an audio stream (Feature Request)

rcarsey1, 2020-08-19:
In my simple case, my server running dump1090/dump978/tar1090 and also has an 3rd SDR plugged in thats receiving VHF audio (airband) which is being fed to a local icecast instance.

The main purpose of my installation is to power a display at my airport's FBO.. However, some people have my tar1090's URL on their tablet/phone and monitor air traffic that way too.  It'd be very convenient to just play the mp3 stream via HTML5 audio controls so that people can watch and hear the planes too as they move around.

My workaround is simply to add some HTML code to play the stream in the background, but I can see how this would be useful in a lot of cases where tar1090 is serving a local airport community.  In a larger/more dynamic scale implementation, icons could be placed on the map in locations where LiveATC has a feed and clicking on them would start playing that particular stream.

rcarsey1, 2020-09-10:
Around line 1024 of index.html (before the closing td tr tags) I added:
`<audio controls preload=none> <source src="http://d.liveatc.net/kewr3_atis" preload=none>
</audio>
`
It was the simplest/convenient place without changing any other code.

ggoodloff, 2023-02-21:
I was able to get this to work....  Nice Add <-  [radar.overbellevue.com] http://radar.overbellevue.com 
The trick was getting the actual feed link.
    Load the webplayer page from liveATC.net
    View the source and search for <audio
    Copy the link of the src:

Open your tar1090 or combine1090 index.html
     search the file for planesTable
     paste the audio link right above this line.  This will look like that:

            </div>

<center>KOMA/KOFF ATC Feed from <a href="https://liveatc.net" target=new"><font color="#ffffff">LiveATC.net</font></a>
<audio controls preload=none src="https://s1-fmt2.liveatc.net/koff3_app?nocache=2023022122002238716" type="audio/mp3"> </audio><center>

            <table id="planesTable" style="width: 100%">