wiedehopf/tar1090 #68: Ability to enter a QNH (pressure value) (Feature request)

jwalatham, 2020-09-14:
It would be useful to be able to add a pressure value (QNH) and Transition Altitude so that aircraft altitudes show correctly, rather than based on 1013.2 all the time. Virtual Radar Server used to have this option in the settings menu, and it was very useful to correct aircraft showing at negative altitudes or "underground".

rcarsey1, 2020-09-15:
I second this.  We overlay tar1090 data on top of a video feed (see [https://www.youtube.com/watch?v=z8lf1nnP4O0](url)).  

Just today, we had a pilot inquire about why it would read -300' as planes were touching down.  We also use this system to help with neighbor noise complaints.  Someone may "complain" that there was an airplane over their house at 500ft (and have the "proof" to back it up!), but in reality, the plane may be at 1000' [pattern altitude] -- try explaining that to an irate airport neighbor!).

I think the fix may be simple: Read an adjustment number from a text file on the webserver (how that number gets there is a job for whomever is setting up tar1090)

peterM, 2022-05-08:
@rcarsey1 you are using these data for what ? You should contact ATS. They have to have radar records stored so they can replay real data correlated from multiple official sources. This is just hobby and could not be used for legal usage.
There is rule that airplane should fly at least 1000ft above houses. BUT Excluding takeoffs and landings. There could be Noise abatement procedures, but they are monitored by ATS and who do not follow usually get fine to pay.
If you have data and you have QNH you can still recalculate manually. How you want to update QNH in system ? Automatically ? Manually ? From where you get official QNH? I mean certified data not just random from internet. Will you update it every 30mins or even more often when there will be SPECI or updated METAR? How you want to decide which QNH to use relating to airdrome? 
So lot of problems and i do not see any benefit. Still these data can not be used for legal usage.

peterM, 2022-05-09:
Ah .. now I see the use-case and start to make sense in my head. I did not know there is possibility to create these overlays.
Maybe there will be better to add some WebHook handlers
-> add custom field (text, number ...) with user defined location and dimensions ,...register handler for this field, ...TAR1090 send data (array of aircraft data) every  xxx seconds (user defined). you will do your math/magic with data and return back to show on map in specified field.

so then you can do what you want with what you want and show where you want.

jorisrobijn, 2022-10-14:
Then why not have auto-QNH, by taking the QNH from the airport nearest to the center of the map?

peterM, 2022-10-14:
Because you can use QNH only under Transition Level (TL) during descend or under Transition Altitude (TA) during climb. Transition altitude is constant over time, but every airport can have different TA.
On other hand TL is not constant and is determined depending on QNH and usually transmitted as part in ATIS broadcast.

So then you have to implement something more robust because you can not use QNH for ACFT in FL 220 because it have to use standard pressure setting and not QNH.

jorisrobijn, 2022-10-14:
I myself am more intersted in lower flying airplanes, not in the ones using flight levels. If auto-QNH were an option, I would use it all the time. 
But is the transition level or transistion altitude not available from the same METAR where you can also get the QNH from? Then you could even automate the display value to change between FL and altitude automaticall where appropriate.

peterM, 2022-10-14:
Transition altitude is defined by Air traffic services. Transition level is calculated by difference between QNH and Standard Pressure 1013,25hPA the way that between have to be at least 1000ft so it is different for QNH higher then STD and different when QNH is lower than STD (simple calc is using ±27ft per hPa). It Is calculated when QNH is measured and updated but on other hand TA is always the same and does not matter on QNH. TA can be different for FIR or even for airports.  In METAR there is QNH, but never TA or TL. METAR is just actual weather report. TA and TL is more local procedures related.

TA is usually in Instrument charts (or AIP) and TL is in ATIS or transmitted by ATC (where ATIS is not available). 
I think the easiest way is to use new column in settings where user can write ICAO code of airport for METAR downloads. Extract QNH from it and recalculate all altitudes.