wiedehopf/tar1090 #70: upintheair.json rings in different colours (feature request)

zenonp, 2020-09-28:
The ability to map heywhatsthat.com upintheair theoretical range is already great, but it could still be improved with different colours for each altitude ring, just like heywhatsthat.com does it. The tag to base it on is "alt":"nnnnn" and comes at the end of each ring definition (there can be any number of altitude rings, e.g. &refraction=0.25&alts=this,that,another,yetanother). For even more perfection, the rings could be coloured according to whatever ColorByAlt is defined in tar1090/html/config.js and the actual ring altitude could be shown in the same way it's shown with the distance rings. 

Admittedly, none of this is functionally necessary and it is only marginally useful. However, it could make a good project for some boring winter evening...

vkirienko, 2020-10-23:
It already works exactly as you describe. Just replace in config.js

//range_outline_colored_by_altitude = false;

with 

range_outline_colored_by_altitude = true;

zenonp, 2020-10-23:
Darn! You're right, it does. Thank you! (It seems wiedehopf already had a boring winter evening ;)

closed 2020-10-23 by zenonp