wiedehopf/tar1090 #71: BLACK ORANGE LINE

Henry2o1o, 2020-10-10:
Just one question, what does the dotted line mean?

[image]

wiedehopf, 2020-10-10:
Too much time between positions means dotted track.

closed 2020-10-10 by wiedehopf

Henry2o1o, 2020-10-11:
I do not understand the lines?

[image]

Henry2o1o, 2020-10-11:
[image]

No balloon flies like this?

wiedehopf, 2020-10-11:
It's MLAT being very inaccurate.
Happens sometimes.

This is not the right place for these questions, please use the adsbexchange forum or discord.

Henry2o1o, 2020-10-11:
ok, thanks