wiedehopf/tar1090 #77: Database load error

rcarsey1, 2020-10-26:
I updated my installation yesterday and again today (the usual command found in the README).  I am getting some errors in the browser.  I also tried deleting /usr/local/share/tar1090 completely and re-downloading everything.  No help.

I am seeing these errors in my Chrome dev mode console:

`a6cb48: Database load error: parsererror at URL: db-a728e0e/A.js
planeObject_a188fb0_a728e0e.js:2002 a02c40: Database load error: parsererror at URL: db-a728e0e/A.js
planeObject_a188fb0_a728e0e.js:2002`

I believe this is somehow related to the "precompress database files in tar1090-db" commit from 4 days ago.  I think maybe I somehow caught things in the middle of switching over to using pre-compressed tar1090-db files.

[https://radar.oldbridgeairport.com](https://radar.oldbridgeairport.com) is my installation for anyone to test.

wiedehopf, 2020-10-26:
Yeah somehow your lighttpd isn't adding the Content-Encoding header properly.

Can you first try this fix:
https://github.com/wiedehopf/tar1090#history-not-loading-issue-possible-fix
(and after the possible fix rerun the tar1090 install script)

wiedehopf, 2020-10-26:
Tempted to revert that change lighttpd .... just too unreliable maybe.

rcarsey1, 2020-10-26:
I understand the issue now.  I MAY be a unique situation, as I'm using nginx, not lighttpd.  (I don't know what is native/defaults are in lighttpd vs nginx vs apache, etc).  I will have to tweak my web server's configs.  I will absolutely agree though that A) there is a lot of savings in compressing these,  B) re-compressing these repeatedly via the web server is wasted compute time.

wiedehopf, 2020-10-26:
The install script produces a custom nginx conf file for each instance.

wiedehopf, 2020-10-26:
```
To configure nginx for tar1090, please add the following line(s) in the server {} section:

include /usr/local/share/tar1090/nginx-tar1090-webroot.conf;
include /usr/local/share/tar1090/nginx-tar1090.conf;
```
does yours not output this?

rcarsey1, 2020-10-26:
Yes, it outputted only the line:
`include /usr/local/share/tar1090/nginx-tar1090.conf;`

I was running on a self-compiled nginx v1.19.2 (i needed to compile some nginx modules for another project).  I checked my old config files (ie.  nginx.conf.rpmsave) and I do see it there.

I uninstalled nginx and installed the CentOS 8 native nginx (v1.14.1).  Fresh config.  Added the include line.

Everything works as intended.

Possible compatibility problem with nginx v1.19..  but its more probable that I didn't compile in some module needed.

Lets chalk this up to user-error.

wiedehopf, 2020-10-26:
I don't think it's the nginx version.
You need to restart nginx when the configuration file for tar1090 changes (doesn't happen too often).

For lighttpd i restart it in the install script but i expect people using nginx to do it themselves ;)
Maybe i should print something extra in that regard.

closed 2020-10-26 by wiedehopf