wiedehopf/tar1090 #83: Track Disappear on page reload/refresh

acebelowzero, 2020-11-04:
Is it possible to keep the track display after a refresh or page reload?
I tried using persistent but it still loses data after a page reload or accessing it the site from a different computer.

wiedehopf, 2020-11-04:
Then the history loading isn't working correctly.
You are talking about a local install?

acebelowzero, 2020-11-04:
I installed tar1090 to my aws server. when I access it from my local computer I can see the track but when I refresh it the page the track and trailing line is gone.

wiedehopf, 2020-11-04:
Check if the tar1090 service is running and you have files in /run/tar1090 ....
Also check the F12 browser console what it says about history loading.

acebelowzero, 2020-11-04:
Before refresh
[image]
After refresh
[image]

Console
[image]

The history is loaded but it seems like its not grabbing the points for the line.

acebelowzero, 2020-11-04:
Also when I go to /persist there no data popping up at all.

wiedehopf, 2020-11-04:
Ah that's on purpose probably because it's a TIS-B track.

wiedehopf, 2020-11-04:
I'd have to check.
Does it do it for all tracks?

acebelowzero, 2020-11-04:
Yes all tracks.

wiedehopf, 2020-11-04:
Link it please.

wiedehopf, 2020-11-04:
If possible.

acebelowzero, 2020-11-04:
Does TIS-B track not save in history?

acebelowzero, 2020-11-04:
I am not sure what you mean to "link it" would you like the link the site?

wiedehopf, 2020-11-04:
Yes.

acebelowzero, 2020-11-04:
Unfortunately this is a private site.
I would you need your public IP to allow only you.

wiedehopf, 2020-11-04:
Message me on the adsbexchange discord then.
https://discord.com/invite/UEXSqDh

closed 2020-11-04 by wiedehopf

wiedehopf, 2020-11-04:
Problem was: field "seen" was missing so tar1090.sh ignored all aircraft.json entries.

dxmekch, 2020-11-16:
Since the latest update I have the same problem

wiedehopf, 2020-11-16:
No you don't have the same problem, you have the same symptom.
Your problem is likely with lighttpd.
Did you try updating just now, if not, do so.

dxmekch, 2020-11-16:
Yes... just reupdated, still not displayed

wiedehopf, 2020-11-16:
As described above i'll need a screenshot of the browser console (F12)

wiedehopf, 2020-11-16:
Then check this:
```
ls /run/tar1090
```

wiedehopf, 2020-11-16:
Or just join this discord and ask in tech-support: https://discord.com/invite/UEXSqDh