wiedehopf/tar1090 #85: Easy Aggregation

marknfischer, 2020-11-09:
I would love to see easy aggregation support added to the config file in master.

RECEIVER_OPTIONS="--net-only --net-connector 192.168.2.7,30005,beast_in"

Is a good model, but getting this to function seems experimental.  Having a model to listen to 'n' endpoints in a clearly documented set of formats as part of the basic deployment seems like an awesome feature, and would instantly allow the average person to replace VRS with ease without having to venture into more esoteric code.

wiedehopf, 2020-11-09:
That would basically mean running an extra readsb --net-only with tar1090.
I've thought about that and i'll probably continue to do just that, only think about it.
We'll see.

You can add the --net-connector option virtually as many times as you like by the way.
You'd usually want to at least grab MLAT results from 30105 (FA) or 30157 (adsbexchange).
Note that if you run readsb to talk to your receiver then specifying --net-only will stop that.

I do have this but it's still using socat and dump1090-fa: https://github.com/wiedehopf/combine1090
It should work nonetheless. (see multiple instances section of tar1090 to make it use /run/combine1090 as a datasource)

Honestly i might just make a separate thing that installs a net-only readsb instance and supplies some config wrapper for readsb to supply the data sources.

marknfischer, 2020-11-09:
In my case, I have all data from all receivers already streaming to an IP address, each on a different port.  So the format you have of RECEIVER_OPTIONS="--net-only --net-connector 192.168.2.7,30005,beast_in" is exactly what I need.  But getting from the basic download to that seems very confusing based on what is online.  To be clear, I am NOT a developer, so I'm depending on good documentation and simplicity.  Your code works brilliantly on a pi running dump1090.  Honestly, I'm loving it.  Making adding aggregation that easy would be a huge win.