wiedehopf/tar1090 #87: config.js option to set range ring text colour to ring colour (Feature Request)

279pb, 2020-11-13:
It would be nice to have a config.js option to set the range ring distance text to the same colour as the circle stroke if set to true, by adding a "fill: new ol.style.Fill({color: circleColor,})," to the 'text' constructor inside the circleStyle in script.js (line 4335). If set to false then leave the default colour.

I have a set of range rings that fade out the further away from the centre you get. The text remains the same colour and (to me) looks out of place. I did this as a quick fix and it looks better (but that's just a personal preference and very subjective!).

Sorry that I'm not yet comfortable with Git and tar0190 to fork, tweak and then send back the changes!

wiedehopf, 2020-11-13:
Yeah that's too esoteric to add even 1 line of code for sorry.
There has to be some limit to the config options.

closed 2020-11-13 by wiedehopf

279pb, 2020-11-13:
Fair enough. Thanks for taking the time to comment. I'll make the tweak to my install when I upgrade my Pi from the master.