wiedehopf/tar1090 #90: Custom Icons appearing invisible but track trace is there.

acebelowzero, 2020-11-29:
I am trying to add custom Icons for certain planes but when doing so the icon does not appear.
I followed the same format as the default icons tar1090 provides but nothing seems to work.

this is my icon:
```
    'aircraft': {
        id: 72,
        w:32,
        h:32,
        viewBox: '0 0 32 32',
        path: 'M191.026 11.811 C 180.009 33.444,174.250 63.958,170.962 118.110 C 169.277 145.866,157.453 163.457,146.559 154.416 C 137.484 146.884,124.508 151.736,123.417 163.069 C 122.552 172.054,71.354 204.601,50.394 209.491 C 22.580 215.979,-5.095 247.704,9.298 256.600 C 10.948 257.619,34.688 252.107,62.054 244.350 C 122.361 227.256,130.704 226.046,130.722 234.388 C 130.729 237.710,132.701 241.084,135.104 241.885 C 138.398 242.983,138.966 241.325,137.413 235.140 C 135.780 228.634,136.339 227.317,140.119 228.767 C 142.738 229.773,144.882 229.026,144.882 227.109 C 144.882 225.141,150.369 223.622,157.480 223.622 L 170.079 223.622 170.554 246.457 C 170.816 259.016,171.382 273.543,171.812 278.740 C 172.243 283.937,173.255 296.746,174.062 307.204 C 174.868 317.663,176.382 329.002,177.425 332.401 C 178.906 337.231,175.888 340.994,163.628 349.606 C 119.146 380.853,120.892 379.070,124.629 389.434 C 128.429 399.970,146.632 400.890,173.779 391.919 C 185.727 387.970,189.687 387.792,194.783 390.975 C 199.815 394.117,202.030 394.094,205.929 390.858 C 209.911 387.554,213.536 387.725,225.870 391.802 C 253.350 400.884,271.555 400.014,275.371 389.434 C 279.108 379.070,280.854 380.853,236.372 349.606 C 224.112 340.994,221.094 337.231,222.575 332.401 C 223.618 329.002,225.132 317.663,225.938 307.204 C 226.745 296.746,227.757 283.937,228.188 278.740 C 228.618 273.543,229.184 259.016,229.446 246.457 L 229.921 223.622 242.520 223.622 C 249.449 223.622,255.118 224.750,255.118 226.129 C 255.118 227.507,262.559 228.863,271.654 229.142 C 280.748 229.420,309.449 235.849,335.433 243.428 C 390.073 259.365,389.362 259.396,387.909 241.130 C 386.874 228.118,370.116 214.275,349.606 209.491 C 328.646 204.601,277.448 172.054,276.583 163.069 C 275.492 151.736,262.516 146.884,253.441 154.416 C 242.547 163.457,230.723 145.866,229.038 118.110 C 224.924 50.345,213.003 3.150,200.000 3.150 C 197.491 3.150,193.452 7.047,191.026 11.811'
    },
```

I added a category for the icon since tar1090 uses categories to determine which icon to use.
`'RAC': ['aircraft', 1],`
The plane object is even logging in console that is has that specific icon.
[image]

This is how it looks on the map
[image]

Any ideas on why this is happening?
Could you give suggestions on icon maker to use if thats the issue?

wiedehopf, 2020-11-29:
disable webGL
or
visit /tar1090/?iconTest and right click save the image.
replace images/sprites002.png with the image you saved.

anyhow for less than 500 planes disabling webGL isn't an issue.

closed 2020-11-29 by wiedehopf

acebelowzero, 2020-11-29:
Does replacing sprites002.png need to be replaced with an image of all the icons I want?
For example If I want to add 10 new icons. Do I put them in one sprite002.png file? or do I make separate
images for each icon? 
Do I also change marker.js code to be as the following?
```
'aircraft': {
        id: 72,
        w:32,
        h:32,
        path: '/images/<icon_name>'
    },
```

acebelowzero, 2020-11-29:
@wiedehopf

wiedehopf, 2020-11-29:
The image is generated by using the browser and going to /tar1090/?iconTest then saving that image.
And it's only one image.
The icons need to be in the places as they are generated.
Actually id 72 will not fit on the current image you need to increase the height of the image so another row is added, see that part of the code in markers.js
```
let glIconSize = 72;
let glImapWidth = 8;                                                                                                                   
let glImapHeight = 9;
```

Really it's easier turning off webGL :)

acebelowzero, 2020-11-29:
I did turn off WebGl but it was still not appearing I even refreshed tar1090.

wiedehopf, 2020-11-29:
Then you have the viewbox wrong.
In inkscape or whichever svg editor you need to have the icon in the viewbox.
Note that the svg editors will often add transformations that are not in the path when you resize, you'd have to resize the path or add the transformation as some icons have it.