wiedehopf/tar1090 #95: Strange: Planes Pointing Wrong Way

memedude, 2020-12-12:
For some strange reason the airplanes on my map are pointing the wrong direction. It makes them appear they are moving backwards.  I installed a new FlightAware Pro Stick USB ADS-B Receiver (Yellow) with external filter and this started happening.

Any ideas of how to fix this strange issue?

Thank You

wiedehopf, 2020-12-12:
That isn't logical and i wouldn't know how tar1090 were to blame.
You should at minimum provide a screenshot via imgur or another site and the text on /tar1090/data/aircraft.json on pastebin.

memedude, 2020-12-12:
For sure, sorry about that.  Here are the 2:

1. https://pastebin.com/VPtaZmp8
1. https://imgur.com/a/5AXmogg

wiedehopf, 2020-12-12:
So ... have you checked /dump1090-fa 

I think it's the underlying program that is the issue not tar1090:
```
{
"hex": "ac8577",
"flight": "AAL1147 ",
"alt_baro": 31000,
"alt_geom": 31750,
"gs": 497.8,
"track": 100.8,
```

That's track 100 degrees being displayed as it should.
In other words, it's not a bug it's intended.

wiedehopf, 2020-12-12:
Oh .. now i noticed you have the map pointing the wrong way.

That's probably just not supported with webGL .... i'd have to check i could probably fix it.
Not sure how many people rotate the map though.

closed 2020-12-12 by wiedehopf

wiedehopf, 2020-12-12:
Should be fixed, added a manual work-around.

memedude, 2020-12-13:
Fixed. I updated to 201213 and it is working great! Thank You