wiedehopf/tar1090 #100: Feature: Device location

Squire298, 2021-02-11:
Hi,

Is there a way to enable a feature that if you are on a device like a mobile phone to include it on the tar1090 map kinda like an overlay?

Perhaps a configuration setting within the config file? 

Cheers

wiedehopf, 2021-02-11:
You mean the local webinterface?
You can only use the device location on websites with https ... so that would be the first obstacle.

Anyhow if the receiver location is known that is displayed as it's the relevant location.
There are no facilities to display a second location.

Squire298, 2021-02-11:
I have my web interface open to tge world, it doesn't seem to be https.

By location was to get location of veiwing device and place it on the msp via its GPS. 

Like Google maps you have the blue dot with the cone on the direction that you are looking. 

Would probably be an extra webgl icon?

wiedehopf, 2021-02-11:
https is something an individual user would have to do.

Anyhow i'm not adding the functionality you're asking for, have a nice day.

closed 2021-02-11 by wiedehopf