wiedehopf/tar1090 #102: togglePersistence vs Trail width

utilisateur47, 2021-02-21:
Hello
Should an aircraft's trail width decrease and increase as Persistence mode is toggled on or off?
Is this by design, I'm not so sure?
For me, this happens regardless of any other option selected.
I couldn't locate any info on this anywhere so I had to ask.
New to GIT so go easy please.
Thank you.
_ut

wiedehopf, 2021-02-21:
That's by design yes as it was originally used manually what ptracks is now.
I suppose i could rewire that.

utilisateur47, 2021-03-01:
Thanks for the reply. 
That's cool, it makes sense to me now.
I'm just a gui user, so was a little unintuitive to me.
Rewire is always an option, is it necessary?
Probably not but that might bug me.
Cheers

closed 2021-03-09 by utilisateur47