wiedehopf/tar1090 #108: CentOS 8

rcarsey1, 2021-03-10:
For whatever its worth, tar1090 performs fine with CentOS 8.  There were two issues I found during install.  One being that apt-get is not used on this platform (I'm not ready to recommend any changes without more testing on my part for this, but in some cases, git is already pre-installed anyway).  

The other is a simple issue: 

https://github.com/wiedehopf/tar1090/blob/8715f22e8aa131ba99ef515065bb85ab63b969d4/install.sh#L24

`--home` and `--quiet` are invalid arguments for my adduser (part of shadow-utils v4.6 on this platform) .   Dropping `--quiet` and changing `--home` to `--home-dir` works OK.

Recommend adding a second `adduser` at Line 26:
```
if ! id -u tar1090 &>/dev/null
then
    adduser --system --home-dir $ipath --no-create-home  tar1090 >/dev/null
fi
```

From my man page:

```
       -d, --home-dir HOME_DIR
           The new user will be created using HOME_DIR as the value for the
           user's login directory. The default is to append the LOGIN name to
           BASE_DIR and use that as the login directory name. The directory
           HOME_DIR does not have to exist but will not be created if it is
           missing.
```

wiedehopf, 2021-03-10:
I've changed the install script to prompt the user to install the packages if apt-get does not exist.
Also added an adduser with alternate syntax.

https://github.com/wiedehopf/tar1090/commit/c4ef56329b292620a1e83f279ee441fe3de7b6da

rcarsey1, 2021-03-10:
Looks good.    Need to remove `--quiet` on the second `adduser` command you added.. else it fails in CentOS 8.   

Not really sure why that flag isn't available on this platform, but adduser is pretty quiet anyway when it succeeds.

wiedehopf, 2021-03-10:
done.

closed 2021-03-10 by wiedehopf