wiedehopf/tar1090 #109: Russian Military

jamesBNR, 2021-03-19:
Per a twitter [thread](https://twitter.com/vcdgf555/status/1372595930860376065) you'll add foreign militaries if you have the hex. I don't know which militaries you are needing. Here's a Russia:

var foreignMilitary = {
    
    "RA-08832": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142280"
    },
    "RA-09341": {
        "Registrant": "Russian Air Force",
        "icaoHex": "14247d"
    },
    "RA-09344": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142480"
    },
    "RA-11037": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142b1d"
    },
    "RA-11178": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142baa"
    },
    "RA-11240": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142be8"
    },
    "RA-11266": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c02"
    },
    "RA-11279": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c0f"
    },
    "RA-11287": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c17"
    },
    "RA-11290": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c1a"
    },
    "RA-11343": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c4f"
    },
    "RA-11344": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c50"
    },
    "RA-11393": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c81"
    },
    "RA-11412": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c94"
    },
    "RA-11413": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142c95"
    },
    "RA-11425": {
        "Registrant": "Russian Air Force",
        "icaoHex": "142ca1"
    },
    "RA-11431": {
        "Registrant": "Russian Air Force",
[... 1398 lines cut ...]
    "RF-75344": {
        "Registrant": "Russian Navy",
        "icaoHex": "1526d9"
    },
    "RF-85855": {
        "Registrant": "Russian Navy",
        "icaoHex": "154e6e"
    },
    "RF-85856": {
        "Registrant": "Russian Navy",
        "icaoHex": "154E72"
    },
    "RF-94296": {
        "Registrant": "Russian Navy",
        "icaoHex": "1501bc"
    },
    "RA-64051": {
        "Registrant": "Russian Post",
        "icaoHex": "14fa33"
    },
    "RA-64052": {
        "Registrant": "Russian Post",
        "icaoHex": "14fa34"
    }
}

closed 2021-03-19 by jamesBNR

wiedehopf, 2021-03-19:
No i said military ranges.

This isn't a range but a collection of serials.
So this would go into the DB ... let's see if it fits in there without type.

wiedehopf, 2021-03-19:
Yeah so ... you'd need to only send me those which don't have the correct data yet.

Just plug the registration or hex into the search box: https://globe.adsbexchange.com/
If you get the data you have there ... then there is no correction needed.
So .. only corrections to the DB please.

wiedehopf, 2021-03-19:
Thank you for the list it was incorporated into the database :)