wiedehopf/tar1090 #111: Speed Not Showing - Mutability

marknfischer, 2021-03-19:
I have this running with Mutability and "combine1090 data redirection only."  It works perfectly, except speed is not being shown.  I can see speed in the data, but not in the display.  Any ideas?  Example message where speed is dropped:

*8d040056990d6e96387047cfb5c3;
CRC: 000000
DF:17 AA:040056 CA:5 ME:990D6E96387047
 Extended Squitter Airborne velocity over ground, subsonic (19/1)
  ICAO Address:  040056 (Mode S / ADS-B)
  Air/Ground:    airborne
  GNSS delta:    1750 ft
  Heading:       245
  Speed:         405 kt groundspeed
  Vertical rate: -1728 ft/min GNSS

wiedehopf, 2021-03-19:
Don't use mutability ......

I'd have to see your aircraft.json and could probably adapt it.

But really this is not a valid issue, this isn't for mutability.
Build yourself readsb ... https://github.com/wiedehopf/adsb-wiki/wiki/Building-readsb-from-source
Or you're using it with some special SDR?

wiedehopf, 2021-03-19:
```
    if (gs != null)
        this.gs = gs;
    else if (data.speed != null)
        this.gs = data.speed;
```

closed 2021-03-19 by wiedehopf

marknfischer, 2021-03-20:
Solved, your comments were very helpful.  
`const gs = isArray? data[2] : data.gs;`
I'm not sure how effective of a hack it is, but the following now shows speed,...
`const gs = isArray? data[2] : data.speed;`

Wiedehopf, so you have DM method you can share?

wiedehopf, 2021-03-20:
Huh... yeah i was a bit confused why this would help.
But i found it and fixed it.
(your fix would break tar1090 ...)

Again dump1090 isn't really supported, you're missing lots of data.
Running a net-only instance of readsb would be recommended .....