wiedehopf/tar1090 #115: API for JSON data?

lance, 2021-03-23:
Hi - this is not really an issue, per se. But I have a feed running that I would like to pull data from. I looked around on this repo and did not find anything that looks like API documentation. What is the best way to access the JSON data feeding the app? Thanks.

wiedehopf, 2021-03-23:
This does not produce any data, it works with dump1090-fa or readsb.

Example readme for readsb: https://github.com/wiedehopf/readsb/blob/dev/README-json.md

Or are you asking how to get an URL?
Just check your browser which json files it's loading in the dev console network tab.

closed 2021-03-23 by wiedehopf

virtadpt, 2021-03-23:
You can get data from http://your-sensor-host/tar1090/data/aircraft.json as JSON.

wiedehopf, 2021-11-16:
tar1090 is only a web interface .... typically used with decoders readsb / dump1090-fa.
readsb has the --write-json option just the same as dump1090-fa.
Anyhow it'll keep overriding the same file so it's limited in usefulness ....

I'm not quite sure what you want, locally the aircraft.json is located in /run/readsb/aircraft.json ... no need to change that from the default service settings.

virtadpt, 2021-11-16:
That data can be pulled via HTTP from http://your-ADSBx-node/tar1090/data/aircraft.json

No need to write code to access it locally.

2ashishs, 2021-11-16:
@wiedehopf / @virtadpt, basically I want to create a database of planes flying over my location.

Earlier, using dump1090-fa with _--write-json "target-dir"_ option, I got _history**N**.json_ files in the "_target-dir_".

Now with readsb, I would like to save _history**N**.json_ files and combine that data with the aircraft info (aircraft-type, country of origin, etc), available in the tar1090 UI.

I tried adding _--write-json_ option in _JSON_OPTIONS_ of _/etc/default/readsb_, but that did not work.

Thanks.

wiedehopf, 2021-11-16:
The historyN.json aren't useful, they get overwritten constantly after half an hour i believe.
Not sure what you want with them.

Anyhow ... something like that is better solved in whatever script / program your using.
You can build with `dpkg-buildpackage -b --build-profiles=rtlsdr,history --no-sign -d` if you really need those history files ....

virtadpt, 2021-11-17:
Poll the aircraft.json file once a minute or so, deduplicate entries, and add them to your database.  That's what I do.

2ashishs, 2021-11-18:
Thanks @wiedehopf and @virtadpt !