wiedehopf/tar1090 #125: Commit c9610f1a72025ce647184b6b9f0ded30333592e7 broke photos

rparenton, 2021-05-11:
https://github.com/wiedehopf/tar1090/commit/c9610f1a72025ce647184b6b9f0ded30333592e7 appears to have broken photos.  I set a breakpoint and found that the root cause is that `selected.regLoaded` is always false, even if the registration is loaded.

`
regLoaded: false
registration: "N348DN"
`

I'm seeing the same behavior in both Firefox and Chrome Edge.  I'm current with https://github.com/wiedehopf/tar1090/commit/c601a36f9ba100c640b3119a7fc6a23f076594c5, and reverting that commit makes photos work again since it reorders the conditional block.

rparenton, 2021-05-11:
That fixed it.  Thanks for the quick response.

I was indeed getting no photos after that commit, so I'll dig deeper into the DB loading and content-encoding.  Since the aircraft types were being populated, I incorrectly assumed that the DB was loading properly.

closed 2021-05-11 by rparenton

wiedehopf, 2021-05-11:
Yeah if the type loads the DB should be working.

Really not sure why you would have no photos then.
So regloaded will be true for you now for all N registered aircraft as it can be derived from the hexid.

That specific plane you mentioned isn't in the DB ... 

Anyhow .. i didn't notice the issue you mention because for all aircraft in the DB the photo loading worked fine for me.

wiedehopf, 2021-05-12:
So i'm reverting the change as actually the change broke db loading.
Thus we'll have to check what your actual issue is .... instead of changing "regLoaded" because that just checks if database data has been loaded.
It's also set to true if there is not database entry, it just isn't set if it fails, i suppose i could change that.

You really should have "Database load error:" in your F12 console ....
If you don't please describe your setup further.