wiedehopf/tar1090 #129: Set home location

vongyver, 2021-05-26:
Great work.  How do we set the home location?  The home centers around NY, even though I am located in western Canada.  

Thank you!

vongyver, 2021-05-26:
I found it in README.md, so much my bad, please disregard and feel free to mock.

closed 2021-05-26 by vongyver

wiedehopf, 2021-05-26:
it's in the readme, i'm surprised, lots isn't in there.
Check out readsb as well if you want and consider feeding adsbexchange:

https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

vongyver, 2021-05-26:
I added a note hoping to catch you before the response and not waste your
time.  I found it.  I could not find in Wiki and overlooked it in the
config initially.
Got it all fixed up and also installed graphs1090 and autogain1090.  Super
work!!  Kudos too.

I will setup to feed adsexchange as well, started with FlightRadar24 code
on a Pi4.

Thanks