wiedehopf/tar1090 #130: Missing Type Code and Registration in latest Safari on OS X

rice3253, 2021-05-27:
Hello,

The type code are missing on Safari on OS X, a Chrome browser on the same machine shows the type code and picture. Chrome, Safari, and OS X have all been updated to the latest versions.

I am noticing I am getting the following console message that is not present in Chrome: a3a0ed: Database load error: parsererror at URL: db-1620696395/A.js
planeObject_1620696395.js:2084

The "a3a0ed" string
[image]
[image]

 is the Hex ID of the aircraft, and varies with each error message in the JavaScript Console log

wiedehopf, 2021-05-28:
So ... what is the latest version you update to ...?

Also show the full output on the browser console if you're already looking at it.

rice3253, 2021-06-01:
[Console.txt](https://github.com/wiedehopf/tar1090/files/6579958/Console.txt)
I have attached the console log, the version number shows as 1622573508.

wiedehopf, 2021-06-01:
That means you installed a locally modified version if i'm not mistaken?

If that browser can't handle the gzip encoding ... really can't help that.
So you'd have to go to the network tab ... reload the page and check the response headers for one of those database files.
Maybe there are duplicated headers that can cause issues.

It's proably a lighttpd issues .. so check the lighttpd logs if the server reports any issues on startup.

closed 2021-06-05 by wiedehopf