wiedehopf/tar1090 #131: Display UAT from other IP

SoNic1967, 2021-05-29:
After editing the config file, the UAT data from another IP does not get displayed on the tar1090 interface.
This is the content of the config file located at /etc/default/tar1090 :
```
# Time in seconds between snapshots in the track history
INTERVAL=8

# How many snapshots are stored in the track history
HISTORY_SIZE=450

# history duration computes as HISTORY_SIZE x INTERVAL seconds.

# Change to yes to enable UAT/978 display in tar1090
ENABLE_978=yes
# If running dump978-fa on another computer, modify the IP-address as appropriate.
# The URL needs to point at where you would normally find the skyaware978 webinterface
URL_978="http://http://192.168.1.44/skyaware978/"

# 1-9 are valid, lower lvl: less CPU usage, higher level: less network bandwidth used when loading the page
GZIP_LVL=5

# hours of tracks that /?pTracks will show
PTRACKS=8
```

rcarsey1, 2021-05-29:
You said it right there:
`http://http//192.168.1.44/skyaware978/ `

SoNic1967, 2021-05-29:
Doh

closed 2021-05-29 by SoNic1967