wiedehopf/tar1090 #133: Proposal: Add German AWACS (E3A) orbits

olithissen, 2021-06-11:
Hello everybody,

after enabling the UK AWACS Orbits I started digging into the regulations for AWACS orbits over Germany. They are defined in the publication **ENR 5.2 Military Exercise and Training Areas** available at https://www.milais.org/publications.php.

I assembled the German orbit areas and orbits into a GeoJSON file at https://github.com/olithissen/AwacsOrbitsDE and I'd be glad if you can put them to good use.

Kind regards
Oli

mpentler, 2022-12-16:
You should submit a PR if possible so it isn’t as much work for @wiedehopf - clone the repo, make a new branch, add your code and commit, and then submit the PR.

mpentler, 2022-12-16:
I’ve gone ahead and tested this on my own local install and it works well, @olithissen do you want me to take on doing a PR for this? Or can you manage yourself?

mpentler, 2022-12-16:
I've just gone ahead and done it, not using the minimal version though.

olithissen, 2022-12-19:
Thank you for going ahead with this @mpentler. I totally forgot about this and haven't checked my mentions over the weekend, so this comes as a nice surprise. 

Min is simply the minified version as in "no pretty printing"

mpentler, 2022-12-19:
No worries! You inspired me to do other countries. Poland is done and I’ve almost finished NL, just checking a coordinate is correct this week (it looks weird) and then that will be complete too.