wiedehopf/tar1090 #134: Does not start on LG WebOS TV browser

sgofferj, 2021-06-16:
I have an LG WebOS TV and was trying to watch tar1090 in the browser app. The page loads but tar1090 doesn't start. The status bar stays on forever. The current dump1090fa web interface works fine, though.

I'm using the today's latest tag from mikenye/tar1090 docker container. It works fine on all computers and Android devices in the household.

The TV is running WebOS 05.10.05
The browser version is ´´´@23.160.downloadable:submissions/23.160.downloadable.3_@258.downloadable.jcl4tv:submissions/258.downloadable.jcl4tv.3-6-g9300c11c´´´

I'm aware that debugging a closed system like a TV is quite a task, so feel free to close as wontfix. If you decide to investigate, please let me know how I can help.

wiedehopf, 2021-06-16:
Yeah current Chrome / Firefox / Safari browser is pretty much all the test coverage i hope for.
Sometimes i fix something if i can ... but mostly it's just too much of an issue catering for old browser versions.

Despite that ... i wanted to have some better error display for some time ... so
First you need the current tar1090 version, in your docker compose add this to the enviroment section for tar1090:
```
      - UPDATE_TAR1090=true
```
Or just add that as an environment variable for however you start the container.
docker stop tar1090
docker rm tar1090
... then start the container again however you please it should come up with the current version

On the TV use the following addition to the URL:
 /?showerrors

It hopefully will print javascript errors to the screen .....

wiedehopf, 2021-06-16:
This will also work as an URL addtion to show errors:
?jse

sgofferj, 2021-06-16:
The latest version does start now. The TV also did an update - might have something to do with that too. It does show an error which I don't see on my PC or mobile. Now it's waiting for a plane to come by to see if it actually works :D.

[image]
[image]

sgofferj, 2021-06-18:
Starts and works now. The error still shows but it doesn't seem to do any harm. Unless you would like to hunt for that JS error, I think, this issue could be closed.

wiedehopf, 2021-06-18:
So it only works with that extra parameter?

I only added that for debugging purposes ... not to fix anything.
Very strange.

If that's what it takes i can without any parameter add such an error handler.

sgofferj, 2021-06-18:
Oh, no, it works without parameter also. Sorry for being unclear.

wiedehopf, 2021-06-18:
Well if it works, it works, closing :)

closed 2021-06-18 by wiedehopf