wiedehopf/tar1090 #135: Question/Feature: Sound if any/specific aircraft enters view

sgofferj, 2021-06-18:
I haven't found anything in the documentation and a quick look over the source also didn't reveal anything...
Is it possible to have tar1090 play a sound if any aircraft or maybe even a configurable aircraft enters the view?
If not, I would like to suggest this as a feature. I could imagine quite a few use cases, my personal being to land my drone if any rescue helicopter comes close to my location, but also people waiting for loved ones on a plane could use that (when it pings, it's time to start driving to the  airport), etc.

wiedehopf, 2021-06-18:
Beyond the scope of tar1090 in my opinion.

closed 2021-06-18 by wiedehopf

sgofferj, 2021-06-19:
Just to clarify, I meant in the client browser, not on the machine, tar1090 is actually running on.