wiedehopf/tar1090 #141: Question: How to submit missing/wrong DB entries?

sgofferj, 2021-07-09:
As the title says... What's the process?
OH-CAY keeps popping up on my screen and she doesn't seem to be in the DB. Her ICAO hex ID would be "460560". She is a club plane of Jukolan Pilotit and does voluntary wildfire watch and SAR flights.
https://www.planespotters.net/photo/1092071/oh-cay-private-cessna-172p-skyhawk-ii

wiedehopf, 2021-07-09:
adsbexchange discord, there is a channel called registration-db

closed 2021-07-09 by wiedehopf

sgofferj, 2021-07-09:
Oh well... Discord demands a mobile phone number for registration, so nothing for me.

wiedehopf, 2021-07-09:
That should depend on the server, as far as i know you don't need phone verification for our server.

sgofferj, 2021-07-09:
Unfortunately, you need for the discord account.