wiedehopf/tar1090 #145: [Enhancement Request] Add Fourth Label Extension for Registration+Type+Altitude+Speed

rdmcvicar, 2021-07-24:
...or adding the speed to the one that already has the other three.

I find myself toggling around to gather those pieces of information as I watch Fisk arrivals to EAA Air Venture / Oshkosh, WI USA.

[image]

I love this tool and appreciate all the work and consideration.  Thanks!

wiedehopf, 2021-07-24:
Use the table on the right maybe?
Columns there can be customized.

If anything labels would need to be completely customizable.
But i don't really care for that, so unless someone else programs it, not happening.

rdmcvicar, 2021-07-24:
Fair enough; I use the columns plenty, but am certainly taking it a step or two further with this request as to focusing on the aircraft screen.

Thanks again for this super tool, and your (also fast) consideration of changes.  Great work.

closed 2021-07-24 by rdmcvicar