wiedehopf/tar1090 #158: Hiding scale and copyright text from the UI

chvvkumar, 2021-11-23:
Hello guys, 

I am trying to make a small display for tar1090 on my desktop. I managed to tune it to my liking except for the scale display and the map copyright information (photo below). Is there a way to hide both of these?

[image]

wiedehopf, 2021-11-23:
It's not allowed to hide the copyright of the maps.

There are no options for this ... it is removable in the javascript but i won't assist with that.

closed 2021-11-23 by wiedehopf

chvvkumar, 2021-11-23:
> It's not allowed to hide the copyright of the maps.
> 
> There are no options for this ... it is removable in the javascript but i won't assist with that.

Understood on the copyright. What about the scale display?

wiedehopf, 2021-11-23:
I'd have to look at the source where it's added.
You're doing custom stuff, search for it yourself.

wiedehopf, 2021-11-23:
1892             new ol.control.ScaleLine({units: DisplayUnits})                                                                        
script.js

chvvkumar, 2021-11-23:
I did look through the documentation but could not find anything related to waht I was looking for.

Appreciate the reference to the code fragment for scale, that is a good jumping off point.