wiedehopf/tar1090 #159: Webinterface Help

fishbum, 2021-11-29:
I apologize in advance for what is most assuredly a basic issue.  I’m having troubles with the webinterface.  It appears I’m tracking Aircraft, the table on the bottom right has AC data, but they aren’t showing up on the map.  I’ve looked for some documentation on this but haven’t found anything.  I’ll attach a screenshot of what I’m currently seeing.  
[image]

Appreciate any help you could provide.

wiedehopf, 2021-11-29:
location in decoder set wrong.

wiedehopf, 2021-11-29:
or reception is just really bad, to get a position you might need better reception.

rcarsey1, 2021-11-29:
This is peculiar.  the aircraft list, of course inherently has the ICAO hex, but also shows the callsigns and altitudes.. but no squawk codes (and I presume no GPS coordinates).  Even is the decoder's GPS position was not set, these entries should at least display the squawk codes.  I can't imagine the receiver "missing" everyone's squawk and position frames.  speed too?

The Persistence button is pressed.. so I'm thinking the aircraft list on the right is stale anyway.

fishbum, 2021-11-29:
> or reception is just really bad, to get a position you might need better reception.

I’ll try to move the antenna to a better position.  Thanks for the reply!

wiedehopf, 2021-11-30:
Improve the whip antenna:
First place it on a can, the shape and size isn't too terribly important, a large metal surface will work as well:
https://media.discordapp.net/attachments/676004018463571989/741376804303929564/581ec3250ace8ce45f961f2a0b3f6e5b7d7bb5dc_2_460x500.png?width=401&height=436
Cut it to length or unscrew the whip and bend some wire around the threads so that the length of the whip is as described here:
https://media.discordapp.net/attachments/676004018463571989/741376818547916891/8d675f88b554d309a628ac92b32d859a34db0513_2_394x500.png?width=343&height=436

You might also want to run
```
sudo systemctl stop readsb
sudo rtl_test
```
check which tuner you have ..... might be a wrong tuner if it's a no-name SDR.

General setup advice: https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver
Some hardware choices: https://github.com/wiedehopf/adsb-wiki/wiki/adsb-receiver-shopping-list

wiedehopf, 2021-11-30:
FD0012/13 is the wrong tuner, RTL820 is correct.

fishbum, 2021-11-30:
> Improve the whip antenna: First place it on a can, the shape and size isn't too terribly important, a large metal surface will work as well: [...]

Thank you so much for the help!

duan78, 2024-04-15:
Hi, I've exactly the same problem; what do you mean by "location in decoder is set wrong" ? How can I check this ? I would like to investigate since it's not a quatlity reception problem ...
[image]

wiedehopf, 2024-04-15:
This is not a webinterface issue, you gotta know which underlying decoder you're running without that info ... no chance to even point you somewhere.

wiedehopf, 2024-04-15:
That looks like a container thing ... well the location is set wrong.
What is so complicated about me saying the receiver location is set wrong?
I guess receiver location is clearer than decoder location.

Wrong latitude / longitude set!

duan78, 2024-04-15:
You are right i'm using "ultrafeeder" from this tuto : https://sdr-enthusiasts.gitbook.io/ads-b ; thx for your reply, I'll try to find this out or install your tool a proper way.

wiedehopf, 2024-04-15:
Press H on the tar1090 interface and it'll show you which location you've set.
That might help you and correct the error.