wiedehopf/tar1090 #160: Label extensions always show speed in "knots"

joberreiter, 2021-12-27:
### Actual behaviour
Label extension shows always speed in "knots", even if units are set to "metrical" or "imperial".
In the example image below, the speed is shown as "119", where the additional mouse hover popup window shows "221 km/h":
[image]

### Expected behaviour
Label extension display should follow the unit setting.
If a permanent "knots" display is the desired behaviour (regardless of the unit setting), add a "kt" to the displayed value to make it clear.

wiedehopf, 2021-12-27:
Please only file bugs after updating, thank you.

joberreiter, 2021-12-29:
You are right, I'm sorry.
Have run the update last week, but maybe this was a caching issue.

I close the issue, I confirm this is fixed.

closed 2021-12-29 by joberreiter

wiedehopf, 2021-12-29:
No problem, seems i was a bit grumpy 2 days ago :)
Yeah i only recently changed something in the code and thanks for confirming it's fixed.