wiedehopf/tar1090 #168: Country flag of Switzerland is squeezed

by, 2022-03-06:
The original country flag form https://dynamospanish.com/flags/countries/switzerland/ is square (as it should be); at https://github.com/wiedehopf/tar1090/blob/master/html/flags-tiny/Switzerland.png, it is stretched to the rectangular format of the other flags

wiedehopf, 2022-03-06:
https://dynamospanish.com/flags/countries/switzerland/

The tiny flag is the stretched already.
It's not an issue.

closed 2022-03-06 by wiedehopf

wiedehopf, 2022-03-06:
gimp to the rescue

by, 2022-03-07:
Maybe use this one with transparent parts left&right as the entire flag is square:
[image]
.

wiedehopf, 2022-03-07:
no it's fine now.