wiedehopf/tar1090 #175: feed issue plus title tab

jamiemc71, 2022-04-09:
hello. 
can i first say you have done some amazing work here with Tar1090.

i have setup an install of tar1090 (220409) and i have setup a feed using the code 
RECEIVER_OPTIONS="--net-only --net-connector MY IP,30003,sbs_in" and entered into /etc/default/readsb as per the instructions.

i have aircraft showing however the radar is unable to work out if the aircraft is MLAT or ADSB or Mode-S eta eta

could you please share some light on how i go about fixing this please so the radar knows what they are and then i can filter by them as i wish to.

secondly. im trying to change the name on the TAB in the browser, i can get it change but once the blue loading bar disappears it changes back to tar1090 :/ i have changed it in the title of the html but i dont know where its picking the Tar1090 up from that it displays once the loading bar has gone.

sorry for combining the 2 issues i have but it made more sense putting it in one place rather then opening 2 help requests.

i look forward to hearing from you.
many thanks
jamie

wiedehopf, 2022-04-09:
Don't use SBS format, use beast or avr/raw instead.

config.js has options for the title, check that in the github readme.

jamiemc71, 2022-04-09:
hey there.

thank you very much for your help :) i changed to beast on port 30005 and it fired up and it now knows the differance :D

i have looked in the config.js and found the sitename and changed that to what i wanted and removed the // so the radar knows to use it, despite me clearing the cache and changing browsers it doesn't seem to have taken effect is there another config it uses on the first install like a example config kind of thing.

many thanks

wiedehopf, 2022-04-09:
search for title in config.js
you modified teh wrong setting.

jamiemc71, 2022-04-09:
what a complete d**** i am hahahah sorry about that.
i have noticed it doesnt seem to matter what i change in the config nothing seems to change for example:
`
-- Title Settings --------------------------------------
Show number of aircraft and/or messages per second in the page title
PlaneCountInTitle = true;
//MessageRateInTitle = false;
`
removed the // on what i need then refreshed cache cleared eta and it hasn't appeared to have done anything so im beginning to think ithear im being dumb and missing something or it hates me hahah

many thanks

wiedehopf, 2022-04-09:
what URL are you looking at?
which config.js are you editing?

jamiemc71, 2022-04-09:
im looking at http:// MY IP/tar1090

and im editing the config here sudo nano /usr/local/share/tar1090/html/config.js which is under Configuring part 2: the web interface on the github read me

wiedehopf, 2022-04-09:
Pagename variable.

It works fine, no clue what you're doing wrong.
Maybe you're not saving the file.

jamiemc71, 2022-04-09:
ive just changed that pagename and its still not changing i just cant put my finger on why.
its defo saving as i use Ctrl-o and Ctrl-x and even restarted tar and the hole server so im at a loss.

i have uploaded my config to https://mega.nz/file/fhoiHAzS#Y5eE4FHHhd4prHY0FoygDwmAcfRoh-Jt0nlAwBIYCPI
would you mind having a quick look and see if you can see where im being dumb as i dont understand this and cant find any reason why its not working

sorry for being such a pain but your help is very much appreciated

wiedehopf, 2022-04-09:
So what path are you editing. You didn't mention.

Anyhow i likely won't answer further, up to you to figure it out.

wiedehopf, 2022-04-09:
Oh you did mention ... should be the correct file name.

No clue.

Maybe wrong server?
Anyhow good luck.

jamiemc71, 2022-04-09:
the server I'm using is Debian 11 dedicated and the site says it does work on Debian so cant see it being the wrong server speacailly if the site is reading that config file.

thanks for your time and help on this

wiedehopf, 2022-04-09:
Well exit nano, re-open and check if it's still changed :)

wiedehopf, 2022-04-09:
You could also add
console.log('test');

this should show in F12 dev console in the browser.

jamiemc71, 2022-04-09:
i have hahaha and it has indeed changed ive even downloaded it to make sure and the changes have saved. 

sorry dumb qestion where do i add console.log('test'); to? ive never used F12 option

wiedehopf, 2022-04-09:
you add it in config.js

google dev console

jamiemc71, 2022-04-09:
so it turns out on the F12 option its showing an syntax error on the config.js9:6 which is the Show number of aircraft and/or messages per second in the page title which would mean its 100% reading the confog and i can also see the console.log('test'); that i added so its just got even more confusing as to why its reading it but not displaying as its ment to.

the only other error is its not found a file called upintheair.json in jquery3.6.0.min.js.2

i shall reinstall the server and start again and see if anything changes and ill try Debian 10 instead of 1 and see if that makes any difference at all.

many thanks for ya time and help

wiedehopf, 2022-04-09:
you need to fix the syntax error you're editing then.
it wasn't there before you edited it.

wiedehopf, 2022-04-09:
(in the file you're editing ... config.js)

jamiemc71, 2022-04-09:
oh ffs hahahah what a dumba**

so it turns out i through i had to take off the // on the hole section that i wanted to use but its just the 1 line i have to do it for.

for example: i put 
Controls page title, righthand pane when nothing is selected
PageName = "My Site";

 but it should of been
// Controls page title, righthand pane when nothing is selected
PageName = "My Site";

many thanks again

closed 2022-04-09 by wiedehopf