wiedehopf/tar1090 #176: map zoom and aircraft list

jamiemc71, 2022-04-09:
hello there again i swear this will be the last you hear from me haha.

i have read the read me files and scrolled through many files and nothing jumps out at me.

i would like to set the aircraft list to only show the aircraft on the map.

what i mean is if I'm only looking at 3 aircraft on the map zoomed in then id like only the 3 to appear in the list as at the moment if I'm looking at 3 it shows every aircraft in the list.

could you point me in the right place of where to set it to-do as i would like please.

many thanks
jamie

wiedehopf, 2022-04-10:
read the tooltips for the buttons.
it's the V button.

closed 2022-04-10 by wiedehopf

jamiemc71, 2022-04-10:
> read the tooltips for the buttons. it's the V button.

I'm aware it's the V button. But I'm looking at removing the V from the display so I want to enable the option so it's always on by default

wiedehopf, 2022-04-10:
well that's a more accurate issue description.

added an option you can set in config.js an a URL query parameter

jamiemc71, 2022-04-10:
> well that's a more accurate issue description.
> 
> added an option you can set in config.js an a URL query parameter

thank you very much for your help :) i shall give it ago