wiedehopf/tar1090 #177: Question about Wind not available

Andechs75, 2022-04-10:
Hi there,
the Columns Wind D. and Wind (kt) doesn't show anything. Any advice where to look at? Is it an config file issue? I'm using dump-fa by the way.
thx a lot.. 
Volker

wiedehopf, 2022-04-10:
Needs ATC interrogation for IAS / TAS / magn. heading.

Also much more accurate with readsb as a backend but still only available when ATC interrogates certain data.

Andechs75, 2022-04-10:
Well, that answer is over my head ;o)
I have some planes showing wind data (when i click on the plane, on the left "detail window"). If that is the source, then i wonder why it's not displayed in the table. But maybe I'm totally wrong with that assumption.

wiedehopf, 2022-04-10:
Oh, that will be using dump1090 as backend and not readsb.

https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite%3A-ADS-B-receiver/

wiedehopf, 2022-04-10:
You'd have to switch ... i think installing combine1090 would work as well :)
https://github.com/wiedehopf/combine1090
(it brings readsb under the hood)

Andechs75, 2022-04-10:
I see, dump1090-fa is not providing the data you need. Therefore i need that "readsb" instead or additionally...
Well, it seems time for a backup, and a long night to give it a try ;o) (especially with my Pi-not-knowledge...)
thx for the advice

closed 2022-04-10 by Andechs75

wiedehopf, 2022-04-10:
combine1090 will work, it shouldn't cause any issues.
you just have to use /combine1090 then.

you just tell it to get the raw data from dump1090.