wiedehopf/tar1090 #178: Flight number instead of callsign?

mensa84, 2022-04-10:
Hello,

not really any issue, but don't know where else to ask questions about this plugin.

Is it possible to show instead of the callsign the flight number?
Is it also possible to link to FR24 website if I click on a flight detail? I think dump1090 does this, or?

wiedehopf, 2022-04-10:
Flight number isn't broadcast on 1090 MHz. Callsign is.

There are FA links available via a config setting.
FR24 can't be properly linked by hex or registration or callsign.
If you know a technical implementation you are free to make a pull request.

wiedehopf, 2022-04-10:
Oh and of course there are also adsbexchange links available via a config setting if i'm not mistaken.

mensa84, 2022-04-10:
Thank you, I enabled FA links by that command `sudo sed -i -e 's?.*flightawareLinks.*?flightawareLinks = true;?' /usr/local/share/tar1090/html/config.js` and refreshed by F5.

But where will I have to click then exactly to open FA?

wiedehopf, 2022-04-10:
click the reg or something, i don't even remember.

mensa84, 2022-04-10:
Not working :(

closed 2022-04-13 by wiedehopf

mensa84, 2022-04-13:
BTW: Do you know why sometimes I can see an airplane in the list on the right, but not showing up in the map?
I had that for example with a helicopter in the last week.