wiedehopf/tar1090 #179: Clarification on TAT/OAT and winds data source

guidocioni, 2022-04-13:
I'm trying to save the values of TAT/OAT and wind speed/direction shown into the web interface inside a database for further analysis. As these are not saved into the history of `tar1090` (as far as I know) I'm writing a Pyhton script which queries `aircraft.json`, which I believe is the same source of data for `tar1090`, and save only these parameters into a DB.

Unfortunately I noticed that the temperature/winds data shown in `aircraft.json` are usually not in line with what I see in web interface of `tar1090`: usually I see only one aircraft reporting weather data in `aircraft.json`, while in the web interface most of the aircrafts at cruise level have that value. 
I tried to also look into the history files but it seems to be the same...

Am I missing something? Are the weather data sourced from `aircraft.json` or enriched from somewhere else?

wiedehopf, 2022-04-13:
for compatibility with dump1090-fa the javascript can calculate the data itself but the accuracy will be lower as readsb enforces the base data to be close in time.
might be that you're not getting the data at the same time due to bad reception or for some reason.

It works fine for me ... i'd have to look at the raw data, are you able to share your MLAT name on adsbexchange? I could look into the raw data that you're sending that way.

guidocioni, 2022-04-13:
> for compatibility with dump1090-fa the javascript can calculate the data itself but the accuracy will be lower as readsb enforces the base data to be close in time. might be that you're not getting the data at the same time due to bad reception or for some reason.
> 

When you say that it calculates those value you mean that it computes wind according to speed information and temperature based on standard atmosphere and pressure/altitude? 

> It works fine for me ... i'd have to look at the raw data, are you able to share your MLAT name on adsbexchange? I could look into the raw data that you're sending that way.

The user should be `eddhi`

One thing that I didn't test was to read ALL the history files together. Maybe I'll then find the right temperature/winds info there and probably I'm just having no luck as `aircraft.json` is updated quite often.

wiedehopf, 2022-04-13:
Look at the code for the calculations please.
It uses true airspeed / ground speed / magnetic heading / ground track
true heading is calculated using a magnetic declination lookup.
that's all mashed together and you get the wind vector.

TAT / OAT are calculated from TAS / mach number.

Are you even running readsb?
tar1090 javascript calculates the same stuff in the browser but doesn't know when the data points where received so it's inaccurate for turns especially.

History files don't have wind.
Really that's as much as i'm going to write, you'll have to dig into the source.

wiedehopf, 2022-04-13:
Im not sure what you're running ... your data shows me wind for 19 aircraft out of 34 for with position.
The 15 without wind just means you don't have sufficient reception on them.

guidocioni, 2022-04-13:
> Look at the code for the calculations please. It uses true airspeed / ground speed / magnetic heading / ground track true heading is calculated using a magnetic declination lookup. that's all mashed together and you get the wind vector.
> 
> TAT / OAT are calculated from TAS / mach number.

Ok sorry, I thought that wind and temperature informations were directly sent into the ADS-B message. 

> 
> Are you even running readsb? tar1090 javascript calculates the same stuff in the browser but doesn't know when the data points where received so it's inaccurate for turns especially.

I'm not running `readsb`, `tar1090` gets the data automatically from an already running instance of `dump1090-fa`. 
Ok so that may explain the winds in the interface.

> 
> History files don't have wind. Really that's as much as i'm going to write, you'll have to dig into the source.

I've actually seen some temperature and wind information in the `aircraft.json` so I'm not sure where is coming from. 

Anyway this was just a question to get to know more, feel free to ignore it if it's not relevant :)

guidocioni, 2022-04-13:
> Im not sure what you're running ... your data shows me wind for 19 aircraft out of 34 for with position. The 15 without wind just means you don't have sufficient reception on them.

Not sure why, I should get correct data. You can see the running instance here https://hh.guidocioni.it/tar1090/

wiedehopf, 2022-04-13:
That doesn't seem to be readsb as a backend, use it!

https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

closed 2022-04-13 by wiedehopf