wiedehopf/tar1090 #180: Filter by country of registration

orangy, 2022-04-18:
I couldn't find if it's possible to show only planes with a specific country of registration. Help, and even query filters manual doesn't mention it. Thanks!

wiedehopf, 2022-04-18:
There is not such feature.

You can filter by hex id though.

orangy, 2022-04-18:
I don't know much about the aviation systems yet, just learning :) Does hex id somehow encode country?

denhamgithub, 2022-07-04:
http://www.aerotransport.org/html/ICAO_hex_decode.html  for blocks of hex codes by country.  You also will find that the A/Craft Registration numbers are by country.  And also are similar to the radio/TV ID's for that country.

Just stumbled onto to your mid-April post and perhaps you've already found the answers to the Q's you raised.

Otherwise, hope you enjoy this hobby as much as me.

mpentler, 2022-11-21:
[image]
??

Did this get added without any fanfare? I was about to try working on it.

closed 2024-05-11 by wiedehopf