wiedehopf/tar1090 #184: Feature Request or Help with coding

Palmeta, 2022-06-02:
Hi, great project! It could be even better if it was possible to add our own layers. I've tried fiddling with layers_ae43201.js in \usr\local\share\tar1090\html but hava had no luck adding elements to the menu. Is it possible? All menu elements are in the file but tar1090 seems to ignore any changes. Any help on this ?

wiedehopf, 2022-06-02:
I'll put something so you can define layers in config.js i suppose.

wiedehopf, 2022-06-02:
Also you could easily do it if you understand the code, just to complete the answer.

Palmeta, 2022-06-02:
Where should I start?

Palmeta, 2022-06-02:
I see code in layers_ae43201.js (in the same folder as config.js) that should be able to be changed and do what I want but nothing happens if I change it. Is it loaded from somewhere else?

Palmeta, 2022-06-02:
A way to bring something in from kml would be perfect. There is already something there for 
` if (tfrs) {
        world.push(new ol.layer.Vector({
            source: new ol.source.Vector({
                url: 'tfrs.kml',
                format: new ol.format.KML(),
            }),
            name: 'tfr',
            title: 'TFRs',
            type: 'overlay',
            opacity: 0.7,
            visible: true,
            zIndex: 99,
        }));
    }`
Butu that doen't bring anything into the menu either.

wiedehopf, 2022-06-02:
Just made some minor changes.
You can now use add_kml_overlay(url, name, opacity) from config.js

It's important that you don't screw up any syntax and check that in the dev console.

You can also do custom_layers.push(........................); from config.js and it should show under custom layers.

Note that config.js is not copied from the source folder using ./install.sh test but rather the installed version is kept intact (honoring the original config.js when updating)

wiedehopf, 2022-06-03:
Maybe discuss it with other people on the adsbexchange discord.

I'm not gonna look through pages of source code if you just dump the whole file.
Why not just put what you added?
Or link a commit on your fork of this project?

Likely your issue is with the Source type in the OpenLayers API being incompatible with the data source.

Palmeta, 2022-06-03:
Ok sorry, I'm not used to forking on github I didnt khow I could do that.
Thank you.

Palmeta, 2022-06-12:
I'm trying to add a static image overlay to Openlayers Map but it seems the Class is missing.  ol.source.ImageStatic does not exist.
Has it been excluded on purpose?

wiedehopf, 2022-06-12:
yes unnecessary stuff isn't included, it's a reduced form of the library to save on js that needs loading.

pushed a version that has it.

closed 2022-08-12 by wiedehopf